camUrban/PteraSoftware #2: Request: Conda Package Support

austinstover, 2020-08-20:
The Conda package manager allows easy installation of packages and, more importantly,  independent environments for compartmentalization of dependencies. PteraSoftware could benefit greatly from the addition of a conda package, perhaps through conda-forge.

The PteraSoftware README recommends installing PteraSoftware through Anaconda, but only a pip package exists, even though Anaconda recommends against installing pip packages since these can create conflicts with the conda package manager.

**Therefore, PteraSoftware should be released as a conda package**, in addition to or as a replacement of its PyPi release.

camUrban, 2020-08-20:
Hi! Thank you for this feature request. I wasn't aware that Anaconda recommends against installing pip packages. Good catch! I will fix this by creating a conda-forge package. I will also update the README to recommend using the conda-forge package for running PteraSoftware in a conda environment and using the pip package for running PteraSoftware in a standard virtual environment.

Does this sound like a good fix to you?

austinstover, 2020-08-20:
Sounds good.

camUrban, 2020-08-28:
Hi! So I've read into getting Ptera Software supported via Conda Forge. It turns out that it is a rather involved process, involving submitting to a review by the Conda Forge team. Given that Ptera Software is still pre-alpha, and unstable, I don't think it would be a good idea to take this step yet. Therefore, I am going to close this issue.

However, I will still amend the readme to recommend against using the anaconda distribution of Python due to the compatibility issues that you pointed out. Thank you!

closed 2020-08-28 by camUrban