camUrban/PteraSoftware #3: Unnecessary Dependencies

austinstover, 2020-08-20:
Codecov, Black, and Pre-commit, according to the README, are all required dependencies of PteraSoftware, even though these are only development tools and are not used during execution of software by a typical user.

In addition, I am uncertain if Codecov even possesses an open-source license, making its installation and usage ethically dubious for some.

Therefore, **Codecov, Black, and Pre-commit should be removed as dependencies from the general pip package**.

camUrban, 2020-08-20:
Hi! Thanks for this bug report. I will remove these dependencies from the REQUIREMENTS file, and the pip package. However, as I am still using Codecov, Black, and Pre-commit for development, I'll leave them in the credits section of the README.

Does this improvement sound like it would fix the issue? Also, do you think there is an ethical issue with using Codecov for development and citing it in the credits section given that it isn't open source?

austinstover, 2020-08-20:
As long as users aren't forced to download a non-open-source software while using an open-source one, I don't see any issue.

camUrban, 2020-08-20:
Okay! In that case, I will continue using Codecov for development but remove it as a dependency.

camUrban, 2020-08-28:
This has been fixed in v0.2.1!

closed 2020-08-28 by camUrban