camUrban/PteraSoftware #4: Missing 2D Plots

EmaSava, 2021-03-14:
Hi,
I was running the 'unsteady_ring_vortex_lattice_method_solver_variable_example' but I can't see any 2D force/moment graphs ones the simulation ends. On the other hand, I'm allowed to save the animation in the 'example' folder pressing 'q' from the visualizer. Have you ever had any similar trouble? I checked all the necessary requirements and 'testing' is set to 'False' in the code. Thanks a lot, this is an incredibly useful tool!!

camUrban, 2021-03-15:
Hi, I'll look into this right now and get back to you.

camUrban, 2021-03-15:
Hello again! So I think I know what the problem is. The figures are displayed using [figure.show()](https://matplotlib.org/3.1.1/api/_as_gen/matplotlib.pyplot.show.html), which should open a window that displays the plot, but doesn't actually save it.

However, some IDEs handle this differently, For example, I use PyCharm which has an option called "Scientific Mode." This option displays the plots in the IDE itself. Apparently, you can [turn this feature off if you'd like](https://www.jetbrains.com/help/pycharm/matplotlib-support.html#console).

Regardless of the IDE you are using, there's a quick workaround! In output.py, you could call the `pyplot.savefig()` function for each plot you'd like to save. For example, if you wanted to save the moment coefficients figure, you'd insert `moment_coefficients_figure.savefig()` after `moment_coefficients_figure.show()` on line 656.

I'll add this functionality in my next feature release because it's a great idea! Let me know if this fixes your issue and thanks for the feedback!

camUrban, 2021-03-17:
@EmaSava, I'm going to close this issue but feel free to open another one if you're still having problems. Thanks again for the feedback!

closed 2021-03-17 by camUrban