camUrban/PteraSoftware #5: Setup Video

camUrban, 2021-03-17:
We should create a setup tutorial video and link to it in the documentation. This should be geared toward a user who doesn't have Python, an IDE, or Ptera Software installed on their computer yet.

legitishan, 2021-04-16:
Hey, I can help out with this :)

camUrban, 2021-04-21:
@legitishan Let me know if you need any advice on this!

rahulrhdev, 2022-10-08:
Hey Team, is this issue still open?

user-536667, 2022-12-10:
> We should create a setup tutorial video and link to it in the documentation. This should be geared toward a user who doesn't have Python, an IDE, or Ptera Software installed on their computer yet.

Hey camUrban, I was wondering if the video is out yet? I am having a really hard time making heads and tails of it since I have zero experience with python. Anything would work, even a low resolution cam video would suffice. Thank you.

camUrban, 2023-01-05:
@rahulrhdev Yes, the issue is still open! Feel free to help out with it if you've successfully installed the package.

camUrban, 2023-01-05:
@user-536667 Hello! Unfortunately, I haven't had any time to work on the video. However, I am on break for the next few weeks, so I'll try to get to it soon. If you're having specific issues, feel free to open a new issue listing what you've tried and what isn't working. Don't be afraid to ask for help; I was also new to Python not too long ago! :smile:

camUrban, 2023-01-23:
Hey all! I finally got around to making [the setup video](https://www.youtube.com/watch?v=oX8u2ZflJM4) and uploading it to YouTube. @user-536667 I hope this helps you out!

closed 2023-01-23 by camUrban

user-536667, 2023-01-23:
@camUrban Thank you for the video cam, I will look into it and post my queries here if I get stuck somewhere.