camUrban/PteraSoftware #7: Testing

camUrban, 2021-03-17:
We should make sure that all the integration tests compare output against expected results. This means getting rid of all the "test_method_does_not_throw" tests.

We could also implement PyTests or some other methodology with a simpler syntax.

legitishan, 2021-04-16:
Hey, I wanna help out with this issue :)

camUrban, 2021-04-21:
@legitishan Let me know if you need any advice on this!

legitishan, 2021-04-21:
Right, I can't run test_output.py, it throws a "ModuleNotFoundError: No module named 'tests'" error for some reason, The folder does contain a __init__.py file, any clue what might be causing this error?

camUrban, 2021-04-24:
@legitishan Okay thanks for letting me know! I just made a new release (v1.1.2) that overhauled a lot of the package structure. Would you mind trying again with the latest version and let me know if that fixes your error?

biswain2309, 2021-09-28:
Hey! Can I contribute to this issue?

camUrban, 2021-10-03:
@biswain2309 Absolutely! Let me know if you have any issues getting started.

YISHISHEN, 2022-03-21:
Hey there!  Great software and help a lot! Thanks for your contribution!
But I have an issue when trying version2.0, my python always has trouble installing "webp" from requirements. Can you help me with that?
I try a lot to install Webp but always does not work.

YISHISHEN, 2022-03-21:
hey there! Great Software!help me a lot.
But I have some issue when runing example. It said  "ZeroDivisionError:division by zero. "
can you help me with that?