camUrban/PteraSoftware #10: [FEATURE] Add requirements_dev.txt

Antyos, 2021-10-01:
# Problem Statement

Issue #3 saw the removal of development dependencies from the `requirements.txt` file. I agree with this decision, however, I do not agree with limiting their reference to only the README.

## Proposed Solution

Instead, development dependencies should be added to a `requirements_dev.txt` file.

James Murphy from the YouTube channel mCoding did a great video about packages that goes over this very concept (and others). See how he adds the `requirements_dev.txt` here: <https://youtu.be/DhUpxWjOhME?t=515>.

camUrban, 2021-10-03:
@Antyos Thanks for the suggestion, this seems like a good idea! I'll work on implementing it in the next release.

camUrban, 2022-01-18:
This was fixed with the latest commit!

closed 2022-01-18 by camUrban