camUrban/PteraSoftware #13: Modeling UVLM for Flapping wing

bharswami, 2022-01-30:
Hi Cameron. I had a doubt. How do you model flapping in UVLM? Do you just give a roll-rate to the wings or do you also change the geometry of the wing as the dihedral changes? Your help will be greatly appreciated.
Thanks
Bharath

camUrban, 2022-02-03:
Hi @bharswami! Thanks for the question. Wing flapping is modeled via the `movement.WingCrossSectionMovement` object. Check out `unsteady_ring_vortex_lattice_method_solver_variable_example.py` in the example directory. This file contains a heavily documented, step-by-step explanation of how the syntax of a flapping-wing vehicle is constructed.

In short, each wing cross section's movement is defined relative to the cross section closer towards the wing's root. This motion is comprised of sweeping (dihedral angle changes), pitching (twist angle changes), and heaving (sweep angle changes). Each of these three components is defined as an oscillating function (typically sinusoidal) with an amplitude and period set by the user. Let me know if that helps!

bharswami, 2022-02-03:
Thanks Cameron. We consider both the angular rates and position changes or
just the position changes?
Regards
Bharath

bharswami, 2022-02-03:
Can you clarify flapping wing in a gmeet please? I am trying to implement
it in my UVLM solver.

camUrban, 2022-02-03:
Entire wing objects can have motion prescribed to their leading-edge (see the `movement.WingMovement` object). However, this isn't implemented for individual wing cross-sections. I'm going to close this issue because I don't think there's a specific question you have about setting up a simulation. I recommend trying to run the example scripts, and then reading through all of their comments. If you run into problems, please open up a new issue using the bug report template.

I'm afraid I don't have time right now to meet in real-time, but if you are interested in the implementation of the wing movement, I recommend reading `movement.py`, it is also pretty well commented. I hope this helps!

closed 2022-02-03 by camUrban