camUrban/PteraSoftware #19: Create Package Installer

Zach10a, 2022-07-01:
# Creation of a Package Installer

Users with no familiarity of Python or IDE's will be unable to use this software. A fully packaged installer created at this stage will save hassle down the line.

## Proposed Solution

Use of PyInstaller and Inno Setup to create package installer.

camUrban, 2022-08-04:
Hmm, interesting idea! I haven't used package installers before, so I have some questions. First, would this be simpler than installing Ptera Software with PyPI? Right now, I think you can get the latest version of the package and all the dependencies by using the command `pip install pterasoftware`.

Zach10a, 2022-08-16:
Hi Cam,

This would be a single installer file, complete with a GUI, that would enable even the most computer-illiterate of users to install and use the package. It would also enable the software to be downloaded from the "Latest Release" section of the Pterasoftware front page.

Zach10a, 2022-08-29:
Now connected to #6 via #18

camUrban, 2023-01-05:
Closing this issue as it is it will be fixed in the latest major version release.

closed 2023-01-05 by camUrban