camUrban/PteraSoftware #20: [BUG]

designJZ, 2022-08-01:
Hi Cam,  I read your description of this project, it's cool, thanks for your work.
But I try to run your example codes several times  and meet this bug.
# Bug Description

1. Go to 'example_solver = ps.steady_horseshoe_vortex_lattice_method.SteadyHorseshoeVortexLatticeMethodSolver(
    steady_problem=example_problem
)

example_solver.run()'

2. See error in SteadyHorseshoeVortexLatticeMethodSolver ---for airplane in self.airplanes :"'Airplane 'object is not iterable."
3. I print its type is class which define in SteadyHorseshoeVortexLatticeMethodSolver. And try to add a metaclass _iter_ but its not work.

## Expected Behavior

Airplane 'object is iterable. and make this command run
example_solver = ps.steady_horseshoe_vortex_lattice_method.SteadyHorseshoeVortexLatticeMethodSolver(
    steady_problem=example_problem
)

## Screenshots

[image]

camUrban, 2022-08-02:
@designJZ thanks for the bug report! I'll take a look at this now.

camUrban, 2022-08-02:
Hey @designJZ! I'm having some issues reproducing your bug. Are you trying to run one of the scripts in `/examples/`, or did you write a custom script?

If you wrote your own script, could you please comment with all the code? If you are running one of the example scripts, could you let me know which one? Thanks!

I don't think the issue is with the source code because I tried running `/examples/steady_horseshoe_vortex_lattice_method_solver.py`, and it executed correctly. Could you verify that this example also works with your copy of Ptera Software?

designJZ, 2022-08-03:
Hi, Cam

This is my full codes I present last time:
# install the packages
!pip install pterasoftware
!pip install -r requirements.txt

#run your example code on github.
import pterasoftware as ps

example_airplane = ps.geometry.Airplane(
    wings=[
        ps.geometry.Wing(
            symmetric=True,
            wing_cross_sections=[
                ps.geometry.WingCrossSection(
                    airfoil=ps.geometry.Airfoil(name="naca2412",),
                ),
                ps.geometry.WingCrossSection(
                    y_le=5.0, airfoil=ps.geometry.Airfoil(name="naca2412",),
                ),
            ],
        ),
    ],
)

example_operating_point = ps.operating_point.OperatingPoint()

example_problem = ps.problems.SteadyProblem(
   airplanes=example_airplane,
   operating_point=example_operating_point
)

example_solver = ps.steady_horseshoe_vortex_lattice_method.SteadyHorseshoeVortexLatticeMethodSolver(
    steady_problem=example_problem
)

example_solver.run()

ps.output.draw(solver=example_solver, show_delta_pressures=True, show_streamlines=True)

then I met the bug.

I run the codes in google colab.

 Today, I tried running /examples/steady_horseshoe_vortex_lattice_method_solver.py and met this new problem:
[image]

# I look the codes in __init__() in Software/pterasoftware/problems.py  --class SteadyProblem:
[image]
there is airplanes as a argument.

I have no idea why I met this "TypeError: __init__() got an unexpected keyword argument 'airplanes'" when I was verifing that this example.

Looking forward to your advice.

camUrban, 2022-08-03:
Hey man! I managed to make your code run on my end with two changes:

- Replace `airplanes=example_airplane` with `airplanes=[example_airplane]` on line 26. This parameter is a list of airplane objects. You can pass in multiple airplanes to simulate formation flights!
- Replace `show_delta_pressures=True,` with `scalar_type="lift",` on line 36. I've depreciated the parameter `show_delta_pressures`. Now, you can see the force coefficients instead.

I forgot to change the code in the README to reflect these changes. I'll make sure it's fixed with the next release! :smile:

designJZ, 2022-08-04:
Thanks for your help Cam.

With your advice I sovle some bugs, but still can not make it run. 

#Problems was sovled:
1 Replace airplanes with airplane solve the bug:"__init__() got an unexpected keyword argument 'airplane'"

2 Replace airplanes=example_airplane with airplanes=[example_airplane]  sovle the bug: "TypeError: object of type 'Airplane' has no len()"

3 Replace show_delta_pressures=True, with scalar_type="lift", sovle the bug: "TypeError: draw() got an unexpected keyword argument 'show_delta_pressures'"

# my current code is
import pterasoftware as ps

example_airplane = ps.geometry.Airplane(
    wings=[
        ps.geometry.Wing(
            symmetric=True,
            wing_cross_sections=[
                ps.geometry.WingCrossSection(
                    airfoil=ps.geometry.Airfoil(name="naca2412",),
                ),
                ps.geometry.WingCrossSection(
                    y_le=5.0, airfoil=ps.geometry.Airfoil(name="naca2412",),
                ),
            ],
        ),
    ],
)

example_operating_point = ps.operating_point.OperatingPoint()

example_problem = ps.problems.SteadyProblem(
   airplanes=[example_airplane],
   operating_point=example_operating_point
)

example_solver = ps.steady_horseshoe_vortex_lattice_method.SteadyHorseshoeVortexLatticeMethodSolver(
    steady_problem=example_problem
)

example_solver.run()

ps.output.draw(solver=example_solver, scalar_type="lift", show_streamlines=True)

# current problem is:
[image]

the problem looks the same as I tried running /examples/steady_horseshoe_vortex_lattice_method_solver.py 

[image]

camUrban, 2022-08-04:
Hey man! I'm glad to hear that those changes helped a little. I tried running your latest script, and it worked fine for me. Here's the image it produced:
[image]

`steady_horseshoe_vortex_lattice_method_solver.py` is also working on my computer.

The new issue may be that you don't have the latest software version. Can you try redownloading the most recent release? You can find it as a zipped folder [here](https://github.com/camUrban/PteraSoftware/releases/tag/v2.2.1). If you are using PyPI to import it as a package, try running these two commands in your local terminal:

`python.exe -m pip install --upgrade pip`
`pip install --upgrade pterasoftware`

designJZ, 2022-08-07:
Hi, Cam. With your help I successfully run the example and verified steady_horseshoe_vortex_lattice_method_solver.py in local Pycharm. This weekend I am struggle with the pycharm on pip some packages, but luckily it work at the end. So thanks again for your help!
In the next few days, I will dive into your project and try to apply it to calculate the aerodynamic of my customized wings.

camUrban, 2022-08-07:
@designJZ That's excellent news! I'm glad I could help, and good luck with your project. Feel free to open another issue if you run into more challenges.

closed 2022-08-07 by camUrban