camUrban/PteraSoftware #21: [FEATURE]

designJZ, 2022-08-08:
Hi, Cam. Today I went through your geometry.py and operating_point.py and areodynamic.py. Then I succefully customizied Operating Points and geometry following the rules and get the aerodynamic result. But I met problem in the method of "Customizable Aircraft Geometry".

# Problem Statement
In geometry.py, Wings can be defined as a collection of two or more wing cross sections of any dimensions and positions. And wing cross sections can be specified to match the mean camber line of an airfoil. In this way, the planform of wing is determined by a group of airfoils designer selected from the "Airfoil" file. It is good to divide into panel and run the solver.

However, I hope to define the Wings in a different way for more variations of shape. For example, wings is a curve like the insects' wing.

## Proposed Solution
I tried to define the Wings directly. Such as, I input the wing planform which represented as a set of points(x,y), and set the wing thickness of wing as a constent number instead of a set of wing_cross_sections. Is that possible in Pterasoftware?

CGTsLu, 2025-04-15:
I'm also having the same issue, I need to customize the wings like insect wing shape, have you solved this problem yet?

camUrban, 2025-08-12:
Hey @CGTsLu and @designJZ. Sorry for the radio silence. I'm finally getting the time to start work on this project again.

I think I encountered a similar issue when running the simulation in validation\validation_study.py. My workaround was to define the wing with 18 separate wing cross sections, each having only one column of panels. This allowed me to approximate a curved wing shape:

[image]

Give that file a read-through and let me know if my method could be adapted to your use case.

This isn't a perfect fix, as it requires checking for spanwise panel convergence manually. Therefore, I'll keep this feature request open in case anyone wants to help make a more robust solution.

camUrban, 2026-02-24:
I'm going to close this as stale for now. We can reopen it in the future if someone would like to work on it.

closed 2026-02-24 by camUrban