camUrban/PteraSoftware #22: [INSTALLATION ISSUE]

bharswami, 2022-09-24:
# Bug Description

I am getting dependency errors while installing PteraSoftware.

## Reproduction

This is while installing Ptera using "pip install PteraSoftware"

## Screenshots

ERROR: Cannot install -r requirements.txt (line 3) because these package versions have conflicting dependencies.

The conflict is caused by:
    pyvista 0.36.1 depends on vtk
    pyvista 0.36.0 depends on vtk<=9.2.0 and >=8.1.2; python_version < "3.10"
    pyvista 0.35.2 depends on vtk
    pyvista 0.35.1 depends on vtk
    pyvista 0.34.2 depends on vtk
    pyvista 0.34.1 depends on vtk

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

## Desktop

- OS: Windows 10

camUrban, 2023-01-05:
Hi @bharswami! Thanks for the bug report. I just ran `pip install pterasoftware`, and it completed with no errors or warnings. Would you mind checking again that you're still having this issue? But first, be sure that you are using Python 3.8 and that you run the command `python -m pip install --upgrade pip`.

Zach10a, 2023-01-21:
@bharswami I have also followed the steps on a new install as per Cam's instruction and they have worked. Please let me know if you are still having an issue.

bharswami, 2023-01-22:
Hi Cameron Urban
After some upgrades I got it working. Thanks for your mail.