camUrban/PteraSoftware #23: New release

Zach10a, 2023-02-12:
# Problem Statement

Preparing for the release of V1.0.0

## Proposed Solution

I am currently compiling a .exe installer using Inno Setup and PyInstaller.

If you could give me the ability to provide releases this would be a very much streamlined process. Get in touch with me via this or my email if you have any questions, however I will try and provide relevant documentation in case you aren't able to give me this level of access. 

I will update this when there is something relevant! Hope you are well.

Zach10a, 2023-02-12:
https://github.com/Zach10a/PteraSoftware/tree/Release_prep

Zach10a, 2023-02-12:
The setup is now working and can be found in the Output folder of that branch. It has some issues, but now a .exe can fully install this entire package.

Zach10a, 2023-02-12:
Having issues with the test test_unsteady_convergence.py. 

Comes up with the following:

> Simulating:100% |█████████████████████████████████████████████████| Elapsed: 00:01, Remaining: 00:00
> Simulating:100% |█████████████████████████████████████████████████| Elapsed: 00:05, Remaining: 00:00INFO:convergence:				Iteration Time: 5.937 s
> INFO:convergence:				Maximum coefficient change from wake type: 156.52%
> INFO:convergence:				Maximum coefficient change from wake length: 159.41%
> INFO:convergence:				Maximum coefficient change from panel aspect ratio: 76.97%
> Simulating:100% |█████████████████████████████████████████████████| Elapsed: 00:05, Remaining: 00:00INFO:convergence:				Maximum coefficient change from chordwise panels: 6153.52%
> INFO:convergence:			Chordwise panels: 6
> INFO:convergence:				Iteration Number: 96/96
> 
> Simulating:100% |█████████████████████████████████████████████████| Elapsed: 00:10, Remaining: 00:00
> INFO:convergence:				Iteration Time: 10.169 s
> INFO:convergence:				Maximum coefficient change from wake type: 1376.6%
> INFO:convergence:				Maximum coefficient change from wake length: 4878.03%
> INFO:convergence:				Maximum coefficient change from panel aspect ratio: 187.23%
> INFO:convergence:				Maximum coefficient change from chordwise panels: 1100.0%
> INFO:convergence:The analysis did not find a converged mesh.
> 
> Error
> Traceback (most recent call last):
>   File "C:\Users\zacht\PycharmProjects\PteraSoftware\tests\integration\test_unsteady_convergence.py", line 81, in test_unsteady_convergence
>     self.assertTrue(abs(converged_num_chords - num_chords_ans) <= 1)
> TypeError: unsupported operand type(s) for -: 'NoneType' and '

int'

Looks like it cant converge, I'm way out of practice but CFL condition error? Let me know

camUrban, 2023-02-17:
Hey @Zach10a! Yes, I noticed that error as well when I was working on the release branch for v3.0.0.

camUrban, 2023-02-17:
Also, I don't see a folder called output in that branch. Would you mind giving me the exact path of the executable?

camUrban, 2023-02-17:
Regarding the ability to make releases, I think you've definitely you've put in enough work to merit that level of access. My only reservation is communicating the specific process I've used in the past. The details are pretty important because I'd like to keep the project's GitHub releases synced with its PyPI releases.

Once we're on the same page with the current implementation, I'd be happy to discuss streamlining/changing the process! Perhaps we should discuss more face-to-face? I'll shoot you an email to see if we can set up a time to chat :)

camUrban, 2023-02-17:
@Zach10a Okay, I just realized I don't have your email, haha. Would you mind shooting me a message at camerongurban@gmail.com?

Zach10a, 2023-03-01:
@camUrban Draft release created w/ binary attached. Looks like it may be ready to merge into Master and get this show on the road!