camUrban/PteraSoftware #28: [FEATURE] Aeroelasticity feedback for UVLM

MOLOCH-dev, 2023-08-18:
# Problem Statement

@RaphZufferey and I are working on a Flapping wing robot with flexible wings. Using PteraSoftware, we aim towards "closing the loop" in the UVLM solver by adding aeroelastic structural effects due to wing flexibility. We have a model that describes structural effects of the wing as wing torsion around the leading edge.  
Currently, PteraSoftware predefines the movement of the wing cross-sections, wings and the airplane as a whole before the UVLM solver is called. The overall aim of our solver is to enable redefinition of the wing with dynamic movement parameters (such as wing twist) at every timestep. This will enhance the capabilities of PteraSoftware and introduce the potential framework for adding feedback loops within simulations.
Integrating this aeroelasticity solver in PteraSoftware this would require some changes to the current structuring of the "Movement" class, which would allow for the redefinition of wing movement at every timestep. We have implemented this change and would like to have an open discussion about whether this method seems feasible or if there are any other approaches currently being investigated.

## Proposed Solution

We would like to add the Aeroelasticity Solver as another feature within the [Unsteady Ring Vortex Lattice Solver](pterasoftware/unsteady_ring_vortex_lattice_method.py). The deformation we are trying to express is a "Wing Cross-section Twist" that can be redefined at every timestep. To implement this, I have mirrored the existing logic of the "Movement" class in PteraSoftware. A short diagram explaining my implementation can be found below :
[image]

This solution is stable and convergent upto a certain extent. At large "twist" angles, the solver is unable to keep up with the sudden deformation. So we plan to further add a convergence layer to interpolate the twist till a stable solution is achieved. We will update you on our progress with the same.

To this end, I have created some bugs/issues in PteraSoftware regarding some force outputs that don't make physical sense, #27. I will like to actively work on this issue as well and am open to your ideas/suggestions on this implementation. I will be able to take on some work to implement the changes/additional features.

camUrban, 2023-08-29:
@MOLOCH-dev Sorry for the delay in getting back to you. I've been a little busy with classes starting this semester. Your potential enhancement sounds awesome!

camUrban, 2023-08-29:
Your proposed implementation sounds feasible, and I don't currently know of anyone else working on integrating aeroelasticity into Ptera Software. Given how important aeroelasticity is to the analysis of flapping wings, you will have my full support in implementing this feature.

camUrban, 2023-08-29:
Also, @MOLOCH-dev, if you think it would be helpful, I'm happy to jump on Zoom to discuss further. My email is camerongurban@gmail.com.

MOLOCH-dev, 2023-08-29:
Thank you for your valuable feedback @camUrban. It is great to have your support. I have emailed you for further discussion on zoom. Excited to keep working ahead on this!

MOLOCH-dev, 2023-10-31:
Hello @camUrban , I wanted to just give an update on the work so far in Aeroelastic simulations using PteraSoftware.
[Webp file](https://drive.google.com/file/d/1Ex-xGoQuKqQNGDqpuF-FbduXPJOGkofU/view?usp=sharing)
I have been able to fix the errant torsion angle at the wingtip and generate these results. Really appreciate your feedback and help so far and look forward to continuing the collaboration! Thank you.

camUrban, 2023-10-31:
@MOLOCH-dev That animation looks awesome! I'm curious, is this level of deflection what you and Dr. Zuffery saw in your robot? Congratulations on the breakthrough and all the good work.

camUrban, 2025-08-12:
Hi @MOLOCH-dev! I'm finally getting the time to maintain this project more actively, so I was wondering if there have been any updates on your work incorporating aeroelasticity into Ptera Software. If you got this working, and we were to validate the results, I think it could be an excellent contribution to the project.

camUrban, 2026-02-24:
Hi all. Quick triage pass: this issue is actively being worked on by @JonahJ27. I’m reviewing open issues and closing stale ones. I’ll leave updates on anything still in progress.