camUrban/PteraSoftware #31: [BUG] Divergence in unsteady_static_convergence_example.py

LeonDeligny, 2025-02-03:
# Bug Description

In the example it is specified that the algorithm should converge to :

# Wake state: prescribed
# Wake length: 6 chord lengths
# Panel aspect ratio: 1
# Chordwise panels: 4S

## Reproduction

I ran unsteady_static_convergence_example.py and got 👍 

## Expected Behavior

Please provide a clear and concise description of what you expected to happen.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Desktop

Please complete the following information.

- OS: ''
- Version: ''

## Additional Context

Add any other context about the problem here.

camUrban, 2025-08-12:
Thanks for the bug report! I'll take a look at this.

camUrban, 2025-11-06:
Hi @LeonDeligny, I've verified this is fixed with the updated unsteady convergence function. I haven't packaged this as part of GitHub release/PyPI package yet, but you should be good to go if you are using the updated `main` branch.